ros/genpy #37: Problem with customized message

groundmelon, 2015-09-10:
I meet a bug when using rosbag filter with a customized message. I've located the bug to this function: https://github.com/ros/genpy/blob/indigo-devel/src/genpy/dynamic.py#L81 

``` python
def _gen_dyn_modify_references(py_text, types):
```

I have a customized message like this:
```
my_message_package/Odometry
  std_msgs/uint8 flags
  std_msgs/string info
  nav_msgs/Odometry odom
  ......
```

Unfortunately, as is shown above, my customized message is **my_message_package/Odometry**, while there is a standard **nav_msgs/Odometry**. The package names are different, but the **type** names are the **same**.

The generated .py script is attached:
```
"""autogenerated by genpy from my_message_package/Odometry.msg. Do not edit."""
import sys
python3 = True if sys.hexversion > 0x03000000 else False
import genpy
import struct

class _nav_msgs__Odometry(genpy.Message):
  _md5sum = "94d99f86002b25504a5d3354fa1ad709"
  _type = "my_message_package/Odometry"
  _has_header = False #flag to mark the presence of a Header object
.............
```

The function ``` def _gen_dyn_modify_references(py_text, types)``` just simply substitutes the class name. In my situation, the ```class Odometry(genpy.Message)``` should be substituted with ```_my_message_package__Odometry```, but the **nav_msgs/Odometry** is first touched during iteration, so it is substituted with ```_nav_msgs__Odometry```

In order to solve the problem, I add few lines to tell this function what is the current working package. See this pull request.  That means that after the comment line ```"""autogenerated by genpy from my_message_package/Odometry.msg. Do not edit."""```, now_pkg is **my_message_package**, not nav_msgs/Odometry, such that it would not substitute this Odometry with **_nav_msgs__Odometry**. When the next comment line is detected, the now_pkg will change. The detection is simply a re-expression.

But the solution is not so "elegant". I'd like to see if anyone has more elegant way to fix this.

dirk-thomas, 2015-09-10:
First of all the msg file you posted seems to be invalid. The field types need to be either:

    std_msgs/UInt8 flags
    std_msgs/String info

or:

    uint8 flags
    string info

Can you please post detailed instructions (e.g. commands to run in the shell) what you are trying to do, what the current behavior is and what you expect it to be after your patch?
If I use the first and build the `my_message_package` package the following file gets generated: https://gist.github.com/dirk-thomas/b14b338ff8c820958200

groundmelon, 2015-09-11:
I wrote a sample package in groundmelon/my_message_package. And the README.md shows how to use this package to raise the bug.

Feel free to connect me if you have any questions about it. Thanks for your help.

dirk-thomas, 2015-09-11:
Thank you for providing the detailed example. I will look into it next week.

dirk-thomas, 2015-09-17:
I was able tot reproduce your problem. I also found another case where I was running into the same issue. When opening your bag file with rqt_bag, opening a raw view for the custom Odometry topic and trying to play a message.

Based on that I created a new patch (#39) which does not need to parse the message text to extract the current message information. Can you please give it a try and confirm that it fixes the problem for you too?

groundmelon, 2015-09-18:
Yes, the problem is fixed. Thanks.

PR #39 (2015-09-18): fix handling of message classes with the same name
This PR replaces #38 and fixes #37.
commit d2c73a68 (2015-09-18): fix handling of message classes with the same name (fix #37)
closed 2015-09-18 by groundmelon