ros/genpy #49: Duration.__floordiv__(Duration, int) is ill-defined

eric-wieser, 2016-03-28:
Floor division makes no sense, because floor makes no sense for duration, as it requires some sense of a base unit.

Two possible sets of semantics could be:

* `Duration // int = Duration.from_nsecs(d.to_nsecs() // int)`
* `Duration // int = Duration.from_secs(d.to_secs() // int)`

Unfortunately, the current semantics are neither of these!

dirk-thomas, 2016-03-28:
That is a good point. I would suggest to:
* return `NotImplemented` for `__floordiv__`
* change `__div__` to be the same as `__truediv__`

Not sure if this should go into Kinetic only or also into Indigo / Jade. While it is a change I can't imagine anyone relying on this weird behavior. @ros/ros_team any comments?

wjwwood, 2016-03-28:
Definitely +1 for Kinetic with a migration note.

For Indigo/Jade, I think it is somewhat likely someone has run into this and knowingly or unknowingly worked around this issue. I also think it would be a nasty thing to catch as it will almost certainly not result in an obvious error but rather a logic change in behavior. So I'm -0 for releasing this change into Indigo and Jade.

eric-wieser, 2016-03-28:
I've just realized that my second implementation choice, `Duration.from_nsecs(d.to_nsecs() // int)`, is exactly the same as `__truediv__`, since nsecs are ultimately rounded anyway

eric-wieser, 2016-03-28:
The current behavior is extremely broken, so I'd be in favor of at the very least changing `__div__` to match `__truediv__`, and adding a deprecation warning to `__floordiv__`

```python
> (Duration(3.2) // 2).to_secs()
1.1
```

eric-wieser, 2016-03-31:
More alarmingly, at some point someone decided that this was [intended behaviour](https://github.com/ros/genpy/blob/indigo-devel/test/test_genpy_rostime_truediv.py#L59)

dirk-thomas, 2016-04-18:
I thought about this some more and came up with #58 which contains a rational why I went for only adding a warnings for this.

PR #58 (2016-04-19): warn about using floor division of durations
Fixes #49.

This adds a `RuntimeWarning` for using floor division with integer dividers. Imo it is reasonable to add this warning to all supported distros since it is informing the user that something unexpected is going on. I don't see a use case where this behavior is actually intended. If existing code currently uses this it will result in a warning and the code an easily be updated.

Regarding the question if the class should always implement true division (for all or only newer ROS distros): Python has the same situation when doing integer divi [... cut]
closed 2016-04-19 by dirk-thomas