ros/genpy #56: Time.__nonzero__ is ill-defined

eric-wieser, 2016-04-01:
What's special about 1970? Why is 1970-1-1 00:00 considered to be false, and everything else true?

I'd argue that `__nonzero__` (and consequently, `__bool__`, `is_zero`) should only be implemented on `Duration` objects

dirk-thomas, 2016-07-08:
Changing this now might break existing uses. I am not if anyone uses it or why but maybe it would be best to add a deprecation warning for it first?

eric-wieser, 2016-07-08:
Deprecation warning sounds good

dirk-thomas, 2016-08-05:
If you could create a PR for this it might speed things up. I am not sure when I will find time to implement and test it myself.

eric-wieser, 2016-08-05:
I'll see what I can do

eric-wieser, 2016-08-05:
Should `Time.is_zero` suffer the same fate? This one is at least explicit in what it is doing, but it's still questionable why it should be needed.

tfoote, 2016-08-05:
As much as it's semantically doesn't make sense to consider some random date in 1970 "false". 

However I think it would be very surprising to me to have a datatype that evaluates to zero, return true.

Since you could get something like `bool(Time(0).to_sec()) != bool(Time(0))`. Which breaks the Python [Truth Value Testing](https://docs.python.org/2/library/stdtypes.html) if you consider it a numeric type. And although it is a user defined class, I'd still consider it a numeric type and thus would follow that pattern.

wjwwood, 2016-08-05:
If you consider the Time object to be an encapsulation of time that can be represented as floating point seconds or as uint64 nanoseconds (both since unix epoch), then I think implementing nonzero and bool() makes sense. However, if you think of it as an encapsulation of time that can be converted to other time specs, e.g. on Windows there is [`This structure is a 64-bit value representing the number of 100-nanosecond intervals since January 1, 1601.`](http://stackoverflow.com/questions/10849717/what-is-the-significance-of-january-1-1601), then maybe it doesn't make sense.

In practice, I don't think it would be surprising for these functions to be implemented, and since we almost exclusively offer it as time since unix epoch I think it's fine to leave them. If I were writing this interface from scratch, then I would consider leaving them out.

eric-wieser, 2016-08-05:
> bool(Time(0).to_sec()) != bool(Time(0)). Which breaks the Python Truth Value Testing if you consider it a numeric type

I think we should base the choice off how `datetime.datetime` behaves here:

```python
from datetime import datetime
d = datetime.fromtimestamp(0)

assert bool(d) is True
```

Here's an example of why implicitly evaluating one specific date to 0 is a bad idea:
```
def do_something_at(time=None):
    if not time:
        time = Time.now()
    ...
```
This will silently ignore the argument if by coincidence the time passed is the epoch

This doesn't really say anything about `.is_zero()`

wjwwood, 2016-08-05:
I think `datetime` is a bad equivalent because ROS Time is a timestamp and datetime is a date. A timestamp and date both have reference points, but a date has a very specific reference frame (gregorian calendar) and has to deal with things like leap seconds, whereas a timestamp is absolute time since a reference point. ROS Time is what you'd pass to `datetime.fromtimestamp()`. I believe it is commonly understood that timestamps in ROS reference the unix epoch or simulated time (which also starts at zero). Even the documentation mentions that zero is considered to be a special state that should be considered when writing your code:

http://wiki.ros.org/roscpp/Overview/Time#Time_zero

For your example, I think that is just a common pitfall of Python code. I personally always use `if kwarg is None:` to check if it was passed by the caller or not.

eric-wieser, 2016-08-05:
> For your example, I think that is just a common pitfall of Python code

But that's kind of the point. Should we allow implicit boolifying of a questionably boolifiable object, or force the use to be explicit and write `if time.is_zero()` if they actually meant to check for zero time?

wjwwood, 2016-08-05:
My point is that you should always check if the kwarg `is None` rather than using `not` for this pattern, regardless of whether or not the object you're expecting can be cast to a bool in a reasonable way. You could also pass `0` or `''` as the parameter and get "now" instead. Checking if something is `None` or not using `not` is just not a good idea. So in that since, it doesn't matter if Time has a `__bool__` or not. There are perhaps other examples which better support your case, but in my opinion this isn't one of them.

As I said before, I don't think it needs to be removed, but if I were designing this from scratch I'd probably leave it out. -0 for removing it.

dirk-thomas, 2016-08-08:
To conclude on the discussion: when looking at the `Time` class as a float representation since epoch (which is how it is being [documented](https://github.com/ros/genpy/blob/cdcf03968aa5ecc4869316159ead566645d8cd41/src/genpy/rostime.py#L53-L54)) the function `is_zero()` makes sense. And when viewing the data structure as a number it is the least surprise for a user to be able to evaluate it as a bool in the same way as other numbers.

In cases where you want a default constructed time to be considered `True` you can check a variable against `None` instead:

```
t = None
if t is None:
    print('time not initialized yet')
t = Time()
if t is not None and not t:
    print('time has been initialized but is zero')
t.secs = 1
if t:
    print('time has been initialized and is not zero')
```

Based on this conclusion I don't think there is a strong enough reason to change the current implementation. Therefore I will close this ticket.

closed 2016-08-08 by dirk-thomas