ros/genpy #68: reserved Python keywords in an array of custom message type

DorianScholz, 2016-10-10:
Usually message fields with a name that equals a reserved Python keyword like "from" are renamed to e.g. "from_". Unfortunately this does not seem to work when the keyword name is used in a sub field of an array of custom messages.
This results in a SyntaxError exception when importing the generated Python message wrapper.

The easiest way to reproduce this is to change "genpy/test/msg/TestString.msg" from:
`string data`
to:
`string from`

Regenerate the test messages:
```
./scripts/genmsg_py.py -p genpy -Igenpy:`pwd`/test/msg -o src/genpy/msg `pwd`/test/msg/TestFillEmbedTime.msg
./scripts/genmsg_py.py -p genpy -Igenpy:`pwd`/test/msg -o src/genpy/msg `pwd`/test/msg/TestFillSimple.msg
./scripts/genmsg_py.py -p genpy -Igenpy:`pwd`/test/msg -o src/genpy/msg `pwd`/test/msg/TestManyFields.msg
./scripts/genmsg_py.py -p genpy -Igenpy:`pwd`/test/msg -o src/genpy/msg `pwd`/test/msg/TestMsgArray.msg
./scripts/genmsg_py.py -p genpy -Igenpy:`pwd`/test/msg -o src/genpy/msg `pwd`/test/msg/TestPrimitiveArray.msg
./scripts/genmsg_py.py -p genpy -Igenpy:`pwd`/test/msg -o src/genpy/msg `pwd`/test/msg/TestString.msg
```

And then run the genpy test suite:
```
catkin_make -DCATKIN_ENABLE_TESTING=1
catkin_make run_tests
```

This produces the following error (3 times):
```
ERROR: test_fill_message_args_embed_time (test.test_genpy_message.MessageTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/dorian/src/ros-stacks/catkin/src/genpy/test/test_genpy_message.py", line 185, in test_fill_message_args_embed_time
    from genpy.msg import TestFillEmbedTime
  File "/home/dorian/src/ros-stacks/catkin/src/genpy/src/genpy/msg/__init__.py", line 1, in <module>
    from ._TestFillEmbedTime import *
  File "/home/dorian/src/ros-stacks/catkin/src/genpy/src/genpy/msg/_TestFillEmbedTime.py", line 87
    _x = val1.from
                 ^
SyntaxError: invalid syntax
```

dirk-thomas, 2016-10-13:
Thanks for reporting this. It seems that `make_python_safe` was not invoked for some recursive cases. #69 should address this. Please comment if it fixes your problem.

DorianScholz, 2016-10-13:
Will test this tomorrow. What do you think about adding a test case for this scenario, to prevent it from occurring again?

dirk-thomas, 2016-10-13:
An additional test sounds always good. Can you please create a pull request with it :wink:

DorianScholz, 2016-10-13:
Sure will do ;-)

PR #69 (2016-10-21): use Python safe subfields for nested types
Fixes #68.
closed 2016-10-21 by dirk-thomas