ros/genpy #73: Message hashes don't change with contents

mikepurvis, 2017-02-08:
Eg:

```
$ python
Python 2.7.11 (default, Jan 22 2016, 08:29:18) 
[GCC 4.2.1 Compatible Apple LLVM 7.0.2 (clang-700.1.81)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from std_msgs.msg import Int8
>>> foo = Int8()
>>> foo.data = 1
>>> hash(foo)
275653405
>>> foo.data = 2
>>> hash(foo)
275653405  # Should be different!
>>> bar = Int8()
>>> bar.data = 1
>>> hash(bar)
275876165  # Should be the same as the first one!
```

The default `hash` method supplied with `object` is based on identity rather than fields, hence the above behaviour.

Instead, generated messages should supply a `__hash__` function which hashes the data, possibly as simple as just passing the message's string representation to `hash`.

dirk-thomas, 2017-02-08:
Neither the generated message class nor its base class override the `__hash__` method. If you run the same code with any other Python object the result is the same for me, e.g.:

```
$ python
Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> class Foo(object):
...     def __init__(self):
...         self.bar = None
... 
>>> f = Foo()
>>> f.bar = 1
>>> hash(f)
8751631690301
>>> f.bar = 2
>>> hash(f)
8751631690301
>>> g = Foo()
>>> g.bar = 1
>>> hash(g)
8751631690309
```

mikepurvis, 2017-02-09:
Oh yes, no argument there. This argument is that they _should_, so that message objects may be added to a `set` for de-duplication purposes, etc.

dirk-thomas, 2017-02-09:
I don't think we should change the default semantic of the Python hash function for objects.

mikepurvis, 2017-02-09:
We already do it as a one-off for `Time` and `Duration` objects:

https://github.com/ros/genpy/blob/e176e70764e1682974fe89ba0ac73dca4d0f383a/src/genpy/rostime.py#L127-L131

The proposal here is simply to extend that behaviour to all messages.

Quoting [the documentation](https://docs.python.org/3.5/reference/datamodel.html#object.__hash__):

> The only required property is that objects which compare equal have the same hash value; it is advised to mix together the hash values of the components of the object that also play a part in comparison of objects by packing them into a tuple and hashing the tuple. Example:
>
>    def __hash__(self):
>        return hash((self.name, self.nick, self.color))
```

Given this, the default implementation based on identity seems bizarre.

closed 2017-02-09 by mikepurvis

dirk-thomas, 2017-02-09:
The `Duration` and `Time` classes are considered to be a fancy representation of a float value. That is why they the hash based on the field values.

Considering that the fields are still mutable I agree that having a hash for those might potentially be "risky".