ros/genpy #78: Time message doc/code inconsistency

asmodehn, 2017-03-20:
I just noticed : 

- http://wiki.ros.org/msg says : time | secs/nsecs unsigned 32-bit ints | ros::Time | rospy.Time 
- https://github.com/ros/genpy/blob/kinetic-devel/src/genpy/rostime.py#L191 and https://github.com/ros/genpy/blob/kinetic-devel/src/genpy/rostime.py#L59 which seems to indicate that Time is implemented with **_signed_** integers... at least for python.

So what is it in reality ? should I expect time secs max value to be 2147483647 (signed max 32 bits) or 4294967295 (unsigned max 32 bits).

dirk-thomas, 2017-03-20:
First of all Python doesn't distinguish signed / unsigned or 32 / 64 bit integers. Looking at C++ instead `Time` is defined by `uint32_t sec, nsec` (see https://github.com/ros/roscpp_core/blob/5820d0672cb9d5fcd1c8cb4a159d04ff20115ff3/rostime/include/ros/time.h#L131).

Some tools reading the `_slot_types` determine the boundary of the type using https://github.com/ros/ros/blob/edaa0cf3061edb9db0f796d651a4614bc2ed77ea/core/roslib/src/roslib/msgs.py#L521. I guess it should be `uint32` to match the other language. Changing it might affect behavior in cases where users rely on the current boundaries though.

asmodehn, 2017-03-21:
Oh thanks for the pointer to that method.
I am also working on a package that introspect the `_slot_types` member and deduce the correct min/max values from it. I'm trying to except as early as possible when the value is bad to make finding mistakes easier.

Python doesn't distinguish signed or unsigned, but we don't want a python node sending a value that is too high for C++ node to receive correctly.
By the way 64bits integers are `long` in python2.7, not like 32bits `int`.

I will have a longer look at roslib to understand how it interprets `_slot_types`.
I was also wondering if we shouldn't overload `_slot_types` for `secs` and `nsecs` to `uint32` in for the `Time` class...
I ll keep digging until I am sure I understand how the time/duration values are processed for python. I ll keep you posted.

asmodehn, 2017-03-22:
Looking at https://github.com/ros/ros/blob/edaa0cf3061edb9db0f796d651a4614bc2ed77ea/core/roslib/src/roslib/msgs.py#L521 this is valid for checking constant types, and time is not a supported constant type apparently, so I'm guessing this code actually doesn't apply to this case...

I found the time_msg definition there : https://github.com/ros/ros/blob/edaa0cf3061edb9db0f796d651a4614bc2ed77ea/core/roslib/src/roslib/msgs.py#L696 which is not what is defined in the genpy.Time class :
```
alexv@AlexV-Linux:~$ source /opt/ros/indigo/setup.bash 
alexv@AlexV-Linux:~$ python
Python 2.7.6 (default, Oct 26 2016, 20:30:19) 
[GCC 4.8.4] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import rospy
>>> import genpy
>>> t = rospy.Time()
>>> t
rospy.Time[0]
>>> t.__slots__
[]
>>> inspect.getmro(type(t))
(<class 'rospy.rostime.Time'>, <class 'genpy.rostime.Time'>, <class 'genpy.rostime.TVal'>, <type 'object'>)
>>> t = genpy.Time()
>>> t.__slots__
['secs', 'nsecs']
>>> t._slot_types
['int32', 'int32']
>>> 
```

However this doesn't seem to affect serialisation : 
```
>>> import StringIO.StringIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named StringIO
>>> import stringio.StringIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named stringio.StringIO
>>> from StringIO import StringIO
>>> buff= StringIO()
>>> value=genpy.Time(secs=-2147483647)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genpy/rostime.py", line 205, in __init__
    raise TypeError("time values must be positive")
TypeError: time values must be positive
>>> value=genpy.Time(secs=4294967295)
>>> from std_msgs.msg import Time as TimeMsg
>>> tm = TimeMsg(data=value)
>>> tm.serialize(buff)
>>> buff.getvalue()
'\xff\xff\xff\xff\x00\x00\x00\x00'
>>> serialized = buff.getvalue()
>>> buff.close()
>>> received = TimeMsg()
>>> received.deserialize(serialized)
data: 
  secs: 4294967295
  nsecs:         0
```
Interestingly we also have : 
```
>>> value=genpy.Time(secs=4294967298)
>>> tm = TimeMsg(data=value)
>>> tm.serialize(buff)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/indigo/lib/python2.7/dist-packages/std_msgs/msg/_Time.py", line 55, in serialize
    except struct.error as se: self._check_types(struct.error("%s: '%s' when writing '%s'" % (type(se), str(se), str(locals().get('_x', self)))))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genpy/message.py", line 333, in _check_types
    raise SerializationError(str(exc))
genpy.message.SerializationError: <class 'struct.error'>: ''I' format requires 0 <= number <= 4294967295' when writing 'data: 
  secs: 4294967298
  nsecs:         0'
>>> value._slot_types
['int32', 'int32']

```
So it seems that the serialization treats time fields as `uint32` (format `I` from https://docs.python.org/2/library/struct.html#format-characters) although I'm not quite sure yet why, because they are declared as `int32` as far as I can see... But Looking at https://github.com/ros/genpy/blob/kinetic-devel/src/genpy/generator.py#L110 it seems these are "special" so there is probably a lot of "special" code here and there...

BTW, while looking around I found out that "length" is using unsigned int32 pack structure https://github.com/ros/genpy/blob/kinetic-devel/src/genpy/generator.py#L384 even though it is called `int32_pack`. A bit confusing...

Looking at https://github.com/ros/genpy/blob/kinetic-devel/src/genpy/rostime.py#L59 it seems `__slots__` are redeclared for both `Time` and `Duration`, and that `_slot_types` is there only for introspection purposes (since it is not used anywhere in there).

From here there are two options : 
1. keep digging and find out why a `time` field with 2 `int32` is serialized as `2I` when a `int32` field is serialized as `i`.
2. drop the ball and treat `Header`, `Time` and `Duration` as "special", not following the usual message constructing logic.

I will go for option 2 for my code right now, following the approach in genpy.
If someone get the time to look into option 1 maybe there will be an opportunity to tighten up the code in here. Feel free to close this if there isn't anything you want to follow at this point.

dirk-thomas, 2017-05-22:
I will close this as you suggested since I don't see what should be done right now. Please feel free to comment on the ticket with specific suggests what to change and I am happy reopen it.

closed 2017-05-22 by dirk-thomas