ros/genpy #81: PyPI release ?

asmodehn, 2017-04-05:
How about doing a pypi release of genpy ?

It would allow a user from python to generate ROS message classes by calling `genpy.genpy_main.genmain` directly from any python script (ROS or not), and would make interfacing with ROS from other python frameworks easier, I think...

My current usecase is that I want to run test on generated python messages classes using tox.

Depending on a debian package, that need to be installed on a specific system, and then accessed through a bash script that setup a environment variable, is quite overkill...

If I had genpy as a `pip install`able package, generating messages just before testing would be much easier, and platform independent...

dirk-thomas, 2017-04-05:
We commonly don't release ROS package to PyPI. Mainly because of a user installs them via pip they overlay any Debian package. So even when the Debian package is being updated the user is stuck with the old version installed via PIP.

Also a lot of ROS package won't have a "complete" setup.py file since it is only being used to provide some of the functionality when being used from within CMake. E.g. commonly are not listed since they are already specified in the package manifest.

That being said if the setup.py file of a package is sufficient you can always simply install it using pip and e.g. the url of the git repo.

asmodehn, 2017-04-06:
> We commonly don't release ROS package to PyPI. Mainly because of a user installs them via pip they overlay any Debian package. So even when the Debian package is being updated the user is stuck with the old version installed via PIP.

pip install should always be done in a virtual environment, otherwise user runs a risk to break its system.
Maybe there should be a warning in the ROS docs somewhere ? https://packaging.python.org/installing/#creating-virtual-environments

Anyway as you mentioned the setup.py is not "complete", so I will try to pull the source (and dependencies) directly for my usecase (using pip on the repo, or via git submodules).

Maybe I ll turn it into a pip package myself or integrate it in one...
I ll post here when I get updates on that.

asmodehn, 2017-04-14:
I'm currently managing , using a git submodule, and a `.pth` to redirect to the package source...

But being able to rely on a released pip package would allow me to : 
- automatically get notified when an update gets released.
- automatically run tests to make sure everything still works with the new version.
- make it trivial to test with different versions of genpy if needed.

All that from a virtual environment, not requiring any specific system setup other than python itself.

FYI the only two pip packages needed to generate ROS messages from a `.msg` or `.srv` file from python are `genpy` and `genmsg`.
When dependent message location are not known, then `rospack` might be needed to find them (with a ROS environment setup) but that's about it.

It would be nice to be able to generate message classes from a pure python environment, as it opens some interesting use cases (especially regarding external uses of ROS, or integrating ROS with other systems...).

So I think there would be some value in it, if OSRF eventually decided to release pip packages for `genpy` and `genmsg`.

dirk-thomas, 2017-04-14:
You can achieve most of this by installing the two packages from their git repos as mentioned before. Either by referencing a specific tag or using master. You only have to check manually if a new version is available if you want to use a specific tag. Release for both of these packages are rather rare and even if a new version is being released it rarely contains "crucial" changes but mostly minor improvements.

Therefore I don't think creating (and maintaining) PyPI releases is important enough for OSRF to spend time on. Especially considering the potential problems users could run into using those (overlaying the Debian packages [while not recommended many users will still install them into their system], mixing these Python packages with different ROS distros, etc.).

If you would like to provide pull requests to extend the `setup.py` files to be more "complete" in order to be able to install the packages via PIP I would be happy to review / merge them (even if they would duplicate some of the information from the `package.xml` manifest).

asmodehn, 2017-04-15:
Ah good point. I had given up quickly on the `pip install` way, since setup.py requires catkin_pkg to be preinstalled.
I made PR for using setuptools : 
- https://github.com/ros/genmsg/pull/68
- https://github.com/ros/genpy/pull/82
Indeed I'm not so happy with the duplication of information, but I don't see any other simple way. Let me know what you think.

dirk-thomas, 2017-04-15:
Currently the CI fails. You need to declare the depenency on setuptools in the `package.xml` file in order for it to be available on the buildfarm.

Maybe some information can be extracted by reading the data from the xml instead of duplicating it? That would make it easier to keep up-to-date in the future.

asmodehn, 2017-04-17:
Regarding extracting data from the xml, that might be more code that we would want.
when creating a virtualenv, we get : 
```
mkvritualexv@AlexV-Linux:~$ mkvirtualenv newtestenv
New python executable in /home/alexv/.virtualenvs/newtestenv/bin/python
Installing setuptools, pip, wheel...done.
(newtestenv) alexv@AlexV-Linux:~$ pip list
appdirs (1.4.3)
packaging (16.8)
pip (9.0.1)
pyparsing (2.2.0)
setuptools (35.0.0)
six (1.10.0)
wheel (0.29.0)
(newtestenv) alexv@AlexV-Linux:~$ 
```
And we have to somehow make do with these, otherwise we would get a complex "multi pip command" install instructions.
So I can't use `libxml`, but I could have a try at it with some `pyparsing`...

Another option that we might want to consider is doing it the other way, and create the packagexml from the setup.py, when running catkin, using something like (https://github.com/mikepurvis/catkin_tools_python/blob/master/catkin_tools_python/create_python_package_xmls.py).
I'm not yet sure where to plug that into catkin however...

Also the information that will change the most is probably the `version` field I am guessing ? which maybe make it worth doing this yet another way : having version defined in one place in the package, and then having `setup.py`, `package.xml` and doc and other extracting it from there. There are many ways to do that: https://packaging.python.org/single_source_version/ (I have been using 3. so far but 5. might be good too...)

Any preference ?

dirk-thomas, 2017-04-17:
The `package.xml` needs to be available as a static resource. It can't be generated since it is being parsed my many tools without ever invoking a build.

I am pretty sure we used to parse the XML done time ago (using `ElementTree`?). But I don't think we used it in virtual envs.

asmodehn, 2017-04-17:
Using `pyparsing` to parse xml can be quite tricky to get right. Just to read the xml prolog I need something like : 
```python
import pyparsing

xmlPrologStart = pyparsing.Word('<?xml')
version = pyparsing.Keyword('version') + pyparsing.Word('=', exact=1) + pyparsing.quotedString()
encoding = pyparsing.Keyword('encoding') + pyparsing.Word('=', exact=1) + pyparsing.quotedString()
standalone = pyparsing.Keyword('standalone') + pyparsing.Word('=', exact=1) + pyparsing.quotedString()
xmlPrologEnd = pyparsing.Word('?>')

xmlProlog = xmlPrologStart + version + pyparsing.Optional(encoding) + pyparsing.Optional(standalone) + xmlPrologEnd

# self test
parsed_prolog = xmlProlog.parseString("<?xml version='1.0'?>")
print(parsed_prolog)
```

which gives me : 
```
['<?xml', 'version', '=', "'1.0'", '?>']
```

So I don't think it makes sense to put all that parsing code into the `setup.py`.
Given that we have to rely on these few libraries to be able to run pip with that package, I think we should not try parsing the `package.xml` from the `setup.py`.

So it seems we are stuck with our current data duplication between package.xml and setup.py.

Other debatable points : 
- the `ros_` prefix for package name : I did that because there is already a `genpy` package on PyPI. Not sure about the impact on the catkin build. Not sure if it s better to have it or not.
- the `dependency_link` option and the git dependency are needed because genmsg is not on PyPI, and because we are getting it from a git repo. Although not really clean, we could just not mention the dependency in the setup.py, but somehow document that users need to always have both, or genpy will not work...

dirk-thomas, 2017-04-20:
Have you tried to use the [ElementTree](https://docs.python.org/2/library/xml.etree.elementtree.html) API as mentioned before?

> the ros_ prefix for package name : I did that because there is already a genpy package on PyPI. Not sure about the impact on the catkin build. Not sure if it s better to have it or not.

We can't avoid such conflicts. So adding the prefix sounds like the only option. I guess it is unlikely that anyone will create a package named `ros-genpy` on PyPI in the future.

> the dependency_link option and the git dependency are needed because genmsg is not on PyPI, and because we are getting it from a git repo. Although not really clean, we could just not mention the [...]

Leaving the dependency out of the setup file is fine with me. This is only for users with custom use cases and I think we can expect them to read the docs for it.

asmodehn, 2017-04-21:
Oh right, I didn't realize that was part of python itself (I thought it was in libxml).
I made a new commit in that PR https://github.com/ros/genpy/pull/82

I tried to keep the parsing simple, yet make the user able to write clean code for the setup() call.
Let me know what you think...

dirk-thomas, 2019-08-12:
See https://github.com/ros/catkin/pull/889#issuecomment-520535277

closed 2019-08-12 by dirk-thomas