ros/genpy #91: Integer division of durations

NikolausDemmel, 2018-02-06:
Today I was surprised to find that integer division of `Duration` converts to floats internally, possibly loosing precision. My use case for using `Duration` over "seconds as a float" is to have arithmetic accurate to the ns. This is for example an issue when averaging durations. For example, I have a bag file where I want to manipulate some topics. Some recorded timestamps are from an arduino clock, i.e. in the order of a few thousand seconds. I want to map this to the recorded timestamps (~ current unix time), and for this I am averaging a bunch of time differences that have a large absolute value, but are within milliseconds and I am very much interested in their ns level precision for the avarage.

The following illustrates the issue:

```
In [50]: s = Duration(100000000,1)

In [51]: s
Out[51]: rospy.Duration[100000000000000001]

In [52]: (s+s)/2
Out[52]: genpy.Duration[100000000000000000]

In [53]: (s+s)//2
Out[53]: genpy.Duration[100000000000000000]

In [54]: np.mean([s, s])
Out[54]: genpy.Duration[100000000000000000]
```

The [current implementation](https://github.com/ros/genpy/blob/52270603dd6b401064a1a5389d9c7092033c7d3a/src/genpy/rostime.py#L371-L430) comes from #59 with previous discussion in #49.

Would it make sense to do this arithmetic with int nanoseconds instead float at least when divisor is an integral type?

dirk-thomas, 2018-02-06:
Improving the precision would indeed be good. I don't mind how the arithmetic is happening internally. Please consider to provide a pull request for this enhancement which also includes test coverage for it.

NikolausDemmel, 2018-02-06:
Thanks for declaring interest.

I hope I get time to investigate what exactly would make sense and provide a PR.

dirk-thomas, 2019-08-12:
Closing due to inactivity.

closed 2019-08-12 by dirk-thomas