ros/genpy #97: Undefined names: 'package' and 'name' in generator.py

cclauss, 2018-08-24:
[flake8](http://flake8.pycqa.org) testing of https://github.com/ros/genpy on Python 3.6.3

$ __flake8 . --count --select=E901,E999,F821,F822,F823 --show-source --statistics__
```
./src/genpy/generator.py:743:76: F821 undefined name 'package'
        raise MsgGenerationException("Cannot generate .msg for %s/%s: %s"%(package, name, str(e)))
                                                                           ^
./src/genpy/generator.py:743:85: F821 undefined name 'name'
        raise MsgGenerationException("Cannot generate .msg for %s/%s: %s"%(package, name, str(e)))
                                                                                    ^
./src/genpy/message.py:231:35: F821 undefined name 'unicode'
            if type(field_val) == unicode:
                                  ^
./src/genpy/message.py:604:17: F821 undefined name 'reload'
                reload(pypkg)
                ^
4     F821 undefined name 'package'
4
```

dirk-thomas, 2018-08-24:
Thank you for the report.

> ./src/genpy/message.py:231:35: F821 undefined name 'unicode'

This line is in a conditional block and only used by Python 2.

Please consider to provide a pull request to address the other errors.

cclauss, 2018-08-24:
Yes.  That block of code does not follow the Python porting best practice [__use feature detection instead of version detection__](https://docs.python.org/3/howto/pyporting.html#use-feature-detection-instead-of-version-detection) so this comment is required to placate linters.

dirk-thomas, 2018-08-24:
@cclauss Please feel free to refactoring existing code to use best practice and provide a pull request for it.

dirk-thomas, 2019-08-12:
The undefined variables have been addressed in #105.

Closing the remaining part of the ticket due to inactivity.

closed 2019-08-12 by dirk-thomas

cclauss, 2019-08-12:
Confirmed: Flake8 on Python 3 no longer finds _undefined names_ in this codebase.