ros/genpy #112: misleading ROSSerializationException in some cases

flixr, 2019-11-21:
If you e.g. have a message like this:
```
string foo
float64 bar
```
and you try to serialize `{"foo": "foo_string", "bar": "not_a_float"}` you get a misleading exception like
```
<class 'struct.error'>: 'required argument is not a float' when writing 'foo_string'
```
instead of what would be correct:
```
<class 'struct.error'>: 'required argument is not a float' when writing 'not_a_float'
```

This seems to be because the local var `_x` is not set before every new field, but this is printed since https://github.com/ros/genpy/commit/f42fa96e243cdfa4b42a25df79e41f4e59fcd327

The solution could be as simple as always setting `_x` before the next field is serialized.

dirk-thomas, 2019-11-21:
Please add a test for this case. Also some tests are failing due to the change.

flixr, 2019-11-21:
done

PR #113 (2020-01-09): serialization: always set _x var for correct exception msg
should fix #112
commit 4a094eaa (2020-01-09): serialization: always set _x var for correct exception msg (#113)

* add test for #112

the SerializationError message should reference the correct field.

* serialization: always set _x var for correct exception msg

should fix #112
closed 2020-01-09 by dirk-thomas