ros/genpy #130: check_type does not check float

mintar, 2020-08-13:
**Problem:** The `check_type` function doesn't check `float32` or `float64` fields. Here's a comparison with the (correct) `int64` behavior:

```python
>>> import genpy.message

################## Floats ##################

# F1. This correctly passes
>>> genpy.message.check_type('x', 'float64', 42.0)

# F2. This should fail ('foobar' is not serializable as float64); during serialization, a generic SerializationError will be thrown:
# genpy.message.SerializationError: <class 'struct.error'>: 'required argument is not a float' when writing 'header: ...'
>>> genpy.message.check_type('x', 'float64', 'foobar')

# F3. This should fail (although np.float64(42) *is* serializable as float64, so unless the user explicitly calls check_type, serialization will work)
>>> genpy.message.check_type('x', 'float64', np.float64(42.0))

################## Ints ##################

# I1. This correctly passes
>>> genpy.message.check_type('x', 'int64', 42)

# I2. This correctly fails ('foobar' is not serializable as float64); during serialization, check_type throws a specific SerializationError:
# genpy.message.SerializationError: field header.seq must be unsigned integer type
>>> genpy.message.check_type('x', 'int64', 'foobar')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 206, in check_type
    raise SerializationError('field %s must be an integer type'%field_name)
genpy.message.SerializationError: field x must be an integer type

# I3. This correctly fails (although np.int64(42) *is* serializable as int64, so unless the user explicitly calls check_type, serialization will work)
>>> genpy.message.check_type('x', 'int64', np.int64(42))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 206, in check_type
    raise SerializationError('field %s must be an integer type'%field_name)
genpy.message.SerializationError: field x must be an integer type
```

**Solution:** There's a `elif field_type in ['float32', 'float64']: ...` missing about here:

https://github.com/ros/genpy/blob/e27b6fd254eee77d0d1445f6b3d17ebfbb60c81a/src/genpy/message.py#L264

**Discussion:** I believe adding a check for floats would be the correct behavior. However, I don't believe that many people use the `check_type` function directly. The normal workflow for most users is this:

* user creates message object and fills fields
* user calls `publish()`, which causes the message to be serialized
* if serialization fails, `check_type` is automatically called to print a specific error message.

The impact of this bug in this scenario is only that `check_type` throws a very unspecific error for floats, while it does print a specific error for all other types.

A different scenario would be if a user directly calls `check_type` to determine if a message can be serialized. In that case, this bug doesn't detect type errors in float fields. In other words, currently, not all messages that pass `check_type` can be correctly serialized. On the other hand, `check_type` currently also rejects some messages (with numpy types) that actually *can* be serialized.

If we fix this bug, `check_type` will be consistent between floats and the other types, and it will no longer accept messages that cannot be serialized. However, it will reject even more messages that could be serialized. What do you think? Do you want me to provide a PR for this, or do we leave it as it is? I don't have any strong opinions either way.

dirk-thomas, 2020-08-17:
A pull request would certainly be appreciated. In order to be considered to be merged and released into Noetic it would probably be better to only fail for types which can't be serialized in order to not introduce regression for currently working code.

mintar, 2020-08-20:
> A pull request would certainly be appreciated.

Done: #131

PR #131 (2020-08-31): Add check for float32 and float64 to check_type
Fixes #130.

Specifically, this PR improves the error messages during serialization. Consider the following script:

```python
from io import BytesIO
from geometry_msgs.msg import PoseStamped

msg = PoseStamped()
msg.pose.position.x = '0.0'
buff = BytesIO()
msg.serialize(buff)
```

**Old behavior:**

```
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/geometry_msgs/msg/_PoseStamped.py", line 106, in serialize
    buff.write(_get_struct_7d().pack(_x.pose.position.x, _x.pose.position.y, _x.pose.position.z, _x.po [... cut]
commit 1d9b218f (2020-08-31): Add check for float32 and float64 to check_type (#131)

* Add check for float32 and float64 to check_type

Fixes #130.

* Allow numpy types in check_type(float)

* Avoid numpy dependency
closed 2020-08-31 by dirk-thomas