ros/genpy #132: regression on kinetic due to #127

jorgenfb, 2020-08-26:
The changes in #127 (https://github.com/ros/genpy/blob/kinetic-devel/src/genpy/generator.py#L761) adds the line:
```
codecs.lookup_error("rosmsg").msg_type = self._type
```
to deserialization. On kinetic this line throws LookupException with the message: `unknown error handler name 'rosmsg'`

The issue can be reproduced by using any package that has regenerated its messages since #127. The package where I discovered it was with rosbridge_suite.

This is how to reproduce it:
```
# Boot up the minimal ros image in docker
docker run -it -p 11311:11311 ros:kinetic-ros-core

# Update apt and install ros-kinetic-rosbridge-suite
apt update
apt install ros-kinetic-rosbridge-suite

# Start roscore
roscore
```

On the host computer run `docker ps` to get the container id.

Log into the container and start rosapi (from rosbridge_suite):
```
docker exec -it <CONTAINER ID> bash

# Inside the container
source /opt/ros/kinetic/setup.bash
rosrun rosapi rosapi_node
```

Log into the container and call service:
```
docker exec -it <CONTAINER ID> bash

# Inside the container
source /opt/ros/kinetic/setup.bash
rosservice call /rosapi/get_param what ever
```

The service call will fail when trying to deserialize the message.

This can probably also be done on your local machine, but I choose docker to make sure we have the same environment.

dirk-thomas, 2020-08-26:
The problem is that you are mixing older and newer Debian packages.

After these instructions:

```
docker run -it -p 11311:11311 ros:kinetic-ros-core
apt update
apt install ros-kinetic-rosbridge-suite
```

you don't have the latest version of `genpy` yet. `dpkg -l | grep genpy` shows version 0.6.7 (which doesn't contain #127). But the newly installed version of `ros-kinetic-rosbridge-suite` was generated with the latest version of `genpy` and therefore expect it to be available.

@mikepurvis FYI since you provided the original patch.

Once the Docker image is being updated the problem should go away. Until then you can invoke `apt dist-upgrade` to make sure that all already installed packages are updated - after that you should have version 0.6.14 (which contains #127) and the problem should go away.

mikepurvis, 2020-08-26:
Yeah, we don't have a good story for expressing a reverse dependency like this— like, everything that builds against pkg X needs to then have a >= run-depend on that version of pkg X.

mikepurvis, 2020-08-26:
I suppose we could anticipate this problem and catch the `LookupError` with a no-op; I don't really have strong opinions either way on it.

dirk-thomas, 2020-08-26:
> I don't really have strong opinions either way on it.

Me neither. Expecting users to have an up-to-date installation isn't too much imo. The problem that the `ros:*` Docker images aren't up-to-date is a long standing one (osrf/docker_images#112).

If someone wants to make a PR to make it more defensive I am happy to review. There will be another Kinetic rebuild (due to a `ros_comm` patch version) coming soon so it wouldn't imply a huge overhead if done now-ish. Otherwise it would trigger a huge rebuild and re-download by users which is questionable if that is worth it.

gwendolynbrook, 2020-08-29:
Posting for anyone else who's seen a similar regression for ros-melodic on 18.04.  This changed caused `rostopic echo` and all of our python subscribers to fail "silently" for *any* custom message using this base image: `FROM arm64v8/ubuntu:18.04 as build`, with the following packages installed:

ros-${ROS_DISTRO}-ros-core
ros-${ROS_DISTRO}-angles
ros-${ROS_DISTRO}-bondcpp
ros-${ROS_DISTRO}-pluginlib
ros-${ROS_DISTRO}-robot-state-publisher
ros-${ROS_DISTRO}-roslint
ros-${ROS_DISTRO}-tf
ros-${ROS_DISTRO}-xacro

ros-${ROS_DISTRO}-uuid-msgs
ros-${ROS_DISTRO}-eigen-conversions
ros-${ROS_DISTRO}-tf2-geometry-msgs
ros-${ROS_DISTRO}-cmake-modules

dirk-thomas, 2020-08-29:
@gwendolynbrook did you run an `apt update && apt dist-upgrade` in the Docker container to ensure you have the latest packages? Then you shouldn't get this error.

gwendolynbrook, 2020-08-29:
> @gwendolynbrook did you run an `apt update && apt dist-upgrade` in the Docker container to ensure you have the latest packages? Then you shouldn't get this error.

update, yes; dist-upgrade no.  What package is the dist-upgrade affecting that's a dependency to get this working?

gwendolynbrook, 2020-08-29:
Still grok'ing the generator code, but I'm wondering why this line isn't also guarded by an "if python3":

https://github.com/ros/genpy/pull/127/files#diff-cda8e9d50594619a9ea61bdd3216b820R759

dirk-thomas, 2020-08-31:
@gwendolynbrook Please provide reproducible steps for your case.

> We see the following error when subscribing to any user defined message type. Note that we are still using python2.7, but this generated code is leaking in where it should not be.

While `codecs.lookup_error("rosmsg").msg_type = self._type` isn't necessary (since unused) with Python 2 the custom error handler is registered unconditionally (https://github.com/ros/genpy/pull/127/files#diff-f4626899ff80377c1eaf9d092e00785eR87) and therefore should be available independent of the used Python version.

I still don't see how you can run into this if you have the latest version of the `genpy` package. So in order to determine the severity of #134 please provide reproducible steps. If this is a general problem it needs an immediate release / sync, if it is a usage problem it is only a minor improvement to avoid the overhead when not being used and doesn't justify a new release at this time.

dirk-thomas, 2020-09-02:
@gwendolynbrook Can you please reply to the above comment.

closed 2020-09-03 by dirk-thomas

gwendolynbrook, 2020-09-05:
> @gwendolynbrook Please provide reproducible steps for your case.
>
>
> While `codecs.lookup_error("rosmsg").msg_type = self._type` isn't necessary (since unused) with Python 2 the custom error handler [...]

Here's my attempt to repro: https://github.com/ros/genpy/compare/kinetic-devel...Maidbot:test-serde-issue (tl;dr -- I'm unable to reproduce in a unit test case).

fmessmer, 2020-09-06:
We started to see a similar error during one of our unittests that started to fail recently for both `kinetic` and `melodic` - without any changes to the test itself:
```
<?xml version="1.0" encoding="utf-8"?>
<testsuite errors="1" failures="0" name="unittest.suite.TestSuite" tests="1" time="1.991"><testcase classname="__main__.PythonMoveGroupTest" name="test_validation" time="0.6440"><error type="LookupError">unknown error handler name 'rosmsg'
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/root/target_ws/src/mojin_manipulation/mojin_manipulation_tests/test/python_move_group.py", line 60, in test_validation
    conversions.msg_from_string(plan1_traj, plan1)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/conversions.py", line 53, in msg_from_string
    msg.deserialize(data)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_msgs/msg/_RobotTrajectory.py", line 249, in deserialize
    codecs.lookup_error("rosmsg").msg_type = self._type
</error></testcase><system-out>&lt;![CDATA[
]]&gt;</system-out><system-err>&lt;![CDATA[
]]&gt;</system-err></testsuite>
```

the effected lines are:
```
            plan1 = self.plan(group, rostest1)
            plan1_traj = RobotTrajectory()
            conversions.msg_from_string(plan1_traj, plan1)
```
where one of MoveIt!'s internal `conversion` functions is used: https://github.com/ros-planning/moveit/blob/master/moveit_commander/src/moveit_commander/conversions.py

If indeed #134 fixes this issue, could we please get this released into kinetic and melodic?
If this is a different issue, let me know what other information is helpful to get things fixed...

dirk-thomas, 2020-09-06:
> If this is a different issue, let me know what other information is helpful to get things fixed...

@fmessmer Do you have the latest version of `genpy` installed when you run into this? If no, please make sure to update to the latest version. If yes, please provide reproducible steps.

fmessmer, 2020-09-07:
@dirk-thomas I can confirm that having `genpy` and `moveit_msgs` from source in our upstream_workspace lets our tests pass again...thus, I hope #134 will get released for `kinetic` and `melodic` some time soon
thanks!

dirk-thomas, 2020-09-10:
> I can confirm that having genpy and moveit_msgs from source in our upstream_workspace lets our tests pass again...thus, I hope #134 will get released for kinetic and melodic some time soon
thanks!

@fmessmer The referenced PR is not planned to be released if the already released version is sufficient (simply because a full rebuild requires a lot of resources and all users have to download new binary packages). So please provide more information if the latest available binary package isn't sufficient.

PR #134 (2020-09-03): python3 guard for using rosmsg codec (take 2)
Replaces #133 (since I didn't have permissions to push to the forked branch) and closes #132.

This patch makes the remaining code for the custom codec error handler conditional on the Python version.

@gwendolynbrook @mikepurvis FYI