ros/genpy #136: Generate type hints for ROS messages and services

bonprosoft, 2021-02-20:
I'm sorry if this is a duplicated issue.

---

Hello.
As you may know, Python has the [type hints](https://www.python.org/dev/peps/pep-0484/) system, which allows us to hint the type of objects, and it is available for not only Python3 but Python2.
By adding type hints, we can use some great tools which make our Python development better, e.g., static type checking by [mypy](http://mypy-lang.org/).
I personally add type hints to my code, and if genpy generates type hints for ROS messages and services, that would be much appreciated.

Is there any plan to add type hints to the generate code?
If no one is working on this feature, I would love to work on it.

tfoote, 2021-02-23:
At a high level this would be reasonable to add to the generator. However with Noetic being the last rosdistro we have no target platform to roll out new features like this so even if you implement it, it's likely to be unable to be deployed.

Looking forward a contribution like this would be appreciated for ROS 2 releases. I believe that the right entry point for that would be https://github.com/ros2/rosidl_python

However before you go ahead with this I'd recommend putting together a proposal for community review so that you make sure that you're on a good path before putting in significant efforts.

bonprosoft, 2021-03-11:
Thank you for your reply and I am so sorry for my late response.
I understood the situation and will try to find another way to contribute to ROS.
Let me close this issue, thank you!

closed 2021-03-11 by bonprosoft