ros/genpy #140: genpy.message.DeserializationError: unpack requires a string argument of length 76

DeepDuke, 2021-07-15:
I'm using ROS Kinetic， and I try to read a rosbag using python package `rosbag` . But I just got this error when started to read it.
` File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 2340, in read_messages
    yield self.seek_and_read_message_data_record((entry.chunk_pos, entry.offset), raw)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 2490, in seek_and_read_message_data_record
    msg.deserialize(data)
  File "/tmp/genpy_1LKntB/tmphfDta4.py", line 916, in deserialize
    raise genpy.DeserializationError(e)  # most likely buffer underfill
genpy.message.DeserializationError: unpack requires a string argument of length 76
`
` for topic, msg, t in self.bag.read_messages():
            if topic == '/mapserver/refined_pose':
                poses.append(msg)
                t1_vec.append(t)
                cnt1 += 1
            if topic == '/pita/info/obs_map':
                map_area = self.compute_mapping_area(msg)
                self.maps_area_vec.append(map_area)
                t2_vec.append(t)
                cnt2 += 1
            print('cnt1 = ', cnt1, 'cnt2 = ', cnt2)
        self.bag.close()
`

sloretz, 2021-07-22:
Thanks for the issue! Unfortunately [ROS Kinetic is End of Life](https://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021). I recommend upgrading to Melodic, Noetic, or ROS 2.

If you continue to need to solve the issue on Kinetic, I would recommend opening a question on https://answers.ros.org with steps someone can copy/paste to reproduce the issue.

closed 2021-07-22 by sloretz

DeepDuke, 2021-07-23:
I have solved this problem. It's not about genpy. Thank you!

rbed23, 2022-02-02:
@DeepDuke how did you solve this issue?

DeepDuke, 2022-02-05:
> @DeepDuke how did you solve this issue?

I found that I'm writing a bag with multi-thread without any lock. So my bag was broken. 😠